Anonoei/klipper_auto_speed #9: Install error

IronAegis90, 2023-11-24:
Error while executing install.sh. This is being executed on a Raspberry Pi 4b running Raspbian bookworm. I tried the fix for Raspberry Pi's of `sudo apt-get install libatlas-base-dev` found in the troubleshooting link in the error message.

**To Reproduce**
Steps to reproduce the behavior:
1. Cloned the Git repository
2. CD'd into install directory
3. Executed install.sh
4. See error

**Printer (please complete the following information):**
 - Printer: Voron 2.4r2
 - Kinematics  CoreXY
 - [ ] klippy.log attached

**Additional context**
Error Message:
Klipper service found!
Linking auto speed to Klipper...
Installing matplotlib in klippy...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting matplotlib
  Using cached matplotlib-3.8.2.tar.gz (35.9 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [122 lines of output]
      /tmp/pip-build-env-z2j9pwic/overlay/lib/python3.11/site-packages/setuptools_scm/git.py:308: UserWarning: git archive did not support describe output
        warnings.warn("git archive did not support describe output")
      /tmp/pip-build-env-z2j9pwic/overlay/lib/python3.11/site-packages/setuptools_scm/git.py:327: UserWarning: unprocessed git archival found (no export subst applied)
        warnings.warn("unprocessed git archival found (no export subst applied)")

      Edit mplsetup.cfg to change the build options; suppress output with --quiet.

      BUILDING MATPLOTLIB
            python: yes [3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0]]
          platform: yes [linux]
             tests: no  [skipping due to configuration]
            macosx: no  [Mac OS-X only]

      running egg_info
      writing lib/matplotlib.egg-info/PKG-INFO
      writing dependency_links to lib/matplotlib.egg-info/dependency_links.txt
      writing requirements to lib/matplotlib.egg-info/requires.txt
      writing top-level names to lib/matplotlib.egg-info/top_level.txt
      Traceback (most recent call last):
        File "/tmp/pip-build-env-z2j9pwic/overlay/lib/python3.11/site-packages/numpy/core/__init__.py", line 24, in <module>
          from . import multiarray
        File "/tmp/pip-build-env-z2j9pwic/overlay/lib/python3.11/site-packages/numpy/core/multiarray.py", line 10, in <module>
          from . import overrides
        File "/tmp/pip-build-env-z2j9pwic/overlay/lib/python3.11/site-packages/numpy/core/overrides.py", line 8, in <module>
          from numpy.core._multiarray_umath import (
      ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

      During handling of the above exception, another exception occurred:

      Traceback (most recent call last):
        File "/tmp/pip-build-env-z2j9pwic/overlay/lib/python3.11/site-packages/numpy/__init__.py", line 130, in <module>
          from numpy.__config__ import show as show_config
        File "/tmp/pip-build-env-z2j9pwic/overlay/lib/python3.11/site-packages/numpy/__config__.py", line 4, in <module>
          from numpy.core._multiarray_umath import (
        File "/tmp/pip-build-env-z2j9pwic/overlay/lib/python3.11/site-packages/numpy/core/__init__.py", line 50, in <module>
          raise ImportError(msg)
      ImportError:

      IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

      Importing the numpy C-extensions failed. This error can happen for
[... 68 lines cut ...]
          self.finalize_options()
        File "<string>", line 80, in finalize_options
        File "<string>", line 80, in <listcomp>
        File "/tmp/pip-install-hc96ce_v/matplotlib_4ab882a6509a4c41818cb512d4cee7ae/setupext.py", line 403, in get_extensions
          add_numpy_flags(ext)
        File "/tmp/pip-install-hc96ce_v/matplotlib_4ab882a6509a4c41818cb512d4cee7ae/setupext.py", line 511, in add_numpy_flags
          import numpy as np
        File "/tmp/pip-build-env-z2j9pwic/overlay/lib/python3.11/site-packages/numpy/__init__.py", line 135, in <module>
          raise ImportError(msg) from e
      ImportError: Error importing numpy: you should not try to import numpy from
              its source directory; please exit the numpy source tree, and relaunch
              your python interpreter from there.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

[notice] A new release of pip is available: 23.3 -> 23.3.1
[notice] To update, run: python -m pip install --upgrade pip

Anonoei, 2023-11-24:
@IronAegis90 
run `sudo apt install python3-numpy` and try it again. I thought I had that troubleshooting step written somewhere in the README. Let me know if that fixes it for you!

IronAegis90, 2023-11-24:
Sorry, I forgot to mention that I had tried that as well. I reran that command again, successfully, and then ran the install.sh but got the same error.

IronAegis90, 2023-11-24:
I just ran into numpy issues trying to get my ADXL345 setup. I found a thread that mentioned running `sudo apt-get install libopenblas-dev` solved their issues when running `TEST_RESONANCE`. Ran that command and it appears to have solved my issues for ADXL and Auto Speed.

closed 2023-12-16 by Anonoei