Anonoei/klipper_auto_speed #12: [BUG] Naming convention for autograph output uses :

HonestBrothers, 2024-01-07:
Naming convention for the auto_speed_graph outputs time on the graph with in HH:MM:SS format. This may cause issues when transferring to Windows computers.

Anonoei, 2024-01-07:
@HonestBrothers easy fix. What delimiter would be best? I'd probably do HH.MM.SS

HonestBrothers, 2024-01-07:
That would cause issues too. I can't actually get the graph to output. Klipper crashes after making the folder.

I'm not sure making a directory is necessary, since this probably won't get ran that often. But if you think the directory is necessary, you might check to see if AutoGraph directory exists, if not make it, if it does, write the file to the directory. 

. as a delimiter might cause issues too, as Windows reserves those for file extensions. So once it sees the . Windows will think that's a file extension, and require renaming of the file to open. 

Also, I think your naming convention of the directory causes the issue where Klipper crashes. I was able to alleviate that changing the directory naming convention, it just names the directory Autograph_MMDDYY, and then names the file AutoGraph_MMDDYYHHMMSS.png

Or something of the sort.

HonestBrothers, 2024-01-07:
I think something along these lines might work. It'll complain about the filepath already existing, so might need some error handling there. 

.PNG still doesn't output for me with the modifications, however.

            filepath = os.path.join(
                self.results_dir,
                f"AUTO_SPEED_GRAPH"
            )
            self.gcode.respond_info(f"Velocs: {velocs}")
            self.gcode.respond_info(f"Accels: {accels}")
            self.gcode.respond_info(f"AUTO SPEED graph found max accel on {aw.axis} after {perf_counter() - start:.0f}s\nSaving graph to {filepath}")
            os.makedirs(filepath, exist_ok=True)
            plt.savefig("AUTO_SPEED_GRAPH_{dt.datetime.now():%Y-%m-%d_%H_%M_%S}_{aw.axis}.png", bbox_inches='tight')
            plt.close()

Anonoei, 2024-01-08:
@HonestBrothers 
It not outputting the png is very strange. I'm not sure what exactly is causing that. To my knowledge I didn't have it create a folder for each run, but maybe that was also part of the PR I accepted?
Thanks for looking through this and giving great answers and solutions!