Anonoei/klipper_auto_speed #13: Sanity Check if not running on an CoreXY Printer[BUG]

Keichide, 2024-01-07:
I have read before that Auto Speed is not made for CoreXY printers, but I wanted to try it out anyway. 

What it does with a Cartesian printer is that it executes the Z_Tilt_Adjust and then does not home the Z axis normally via G28 Z but instead moves the Z axis completely upwards and waits until it bumps into nothingness, which does not work with a Cartesian for homing the Z axis, as it would have to go downwards. 

I would therefore recommend that before the script does anything at all, you let it look in the config to see what is set for [printer] and only execute a G28 Z accordingly, in my [printer] section for example it cleary starts that the used Kinematics are cartesian.

Anonoei, 2024-01-08:
@Keichide 

I'm not sure what would cause this issue. 
From the code below it looks like I have it do a "G28 Z0" instead of just G28. Could you try running "G28 Z0" and see if it gives you the same result? Maybe I just need to remove that 0. 

def _home(self, x=True, y=True, z=True):
        prevAccel = self.toolhead.max_accel
        prevVeloc = self.toolhead.max_velocity
        self._set_velocity(self.th_veloc, self.th_accel)
        command = ["G28"]
        if x:
            command[-1] += " X0"
        if y:
            command[-1] += " Y0"
        if z:
            command[-1] += " Z0"
        self.gcode._process_commands(command, False)
        self.toolhead.wait_moves()
        self._set_velocity(prevVeloc, prevAccel)

Keichide, 2024-01-09:
No, a G28 z0 does exactly the same thing as the G28 z.

I'm also currently serachring in the Code why he is doing the Z_Tilt_Adjust and if its Optional or not.