Anonoei/klipper_auto_speed #17: Homing Speed and Z High in config [ENHANCE]

Fragmon, 2024-02-28:
Thank you for your efforts.
With your work, I can now conclude my enhancements :-)
[https://github.com/cryd-s/klipper_scripts/tree/main/speed_accel_script](https://github.com/cryd-s/klipper_scripts/tree/main/speed_accel_script). 

I'd like to suggest a couple of enhancements:

1. Could we include an option in the configuration to set a fixed Z height? This would eliminate the need to specify it in the command line for each test.
2. Is it possible to adjust the homing speed? The script executes numerous homing actions, and since my setup has a high homing speed, a parameter to lower the homing speed during the script's execution would be beneficial.

For future developments, I propose the following feature. In my macro, I've experimented with incrementally increasing the current when the motor skips. While somewhat risky, this could potentially enable us to push the performance boundaries further. My macro tested the maximum acceleration and speed, and in the event of a stepper motor failure, it would slightly increase the current.

Anonoei, 2024-02-28:
@Fragmon 
Firstly, I only decided to learn how to make klipper plugins because I couldn't figure out how to get stepper steps to do automated skipping detection. You did an awesome job with those macros!

1. I can add in a configuration option for the Z height. 
2. I'm not sure. It is probably possible, and I could see how that would be beneficial for getting the best accuracy. Maybe a configuration option for "variable homing speed" and then you can specify the speed you want to use?

Increasing the motor current actually sounds really cool. I imagine people setting goals for what speeds they want their printer to reach, and it automatically tuning the motor currents to reach those speeds. Doing this would also make it easier to test how current effects the steppers, which would be beneficial

For now, I'll add in the Z height configuration, and I'll add the homing speed and motor current to the roadmap. Thank you for your work with that macro, and for sharing your ideas!

Fragmon, 2024-02-28:
> @Fragmon Firstly, I only decided to learn how to make klipper plugins because I couldn't figure out how to get stepper steps to do automated skipping detection. You did an awesome job with those [...]

The idea about the current is in the dev folder. Maybe you can use this.