Anonoei/klipper_auto_speed #18: [BUG] The Z value in the AUTO_SPEED Z=x command is not being recognized

Fragmon, 2024-03-06:
**Describe the bug**
No matter what margin or Z height I set, my printer always moves to 152.5 in the Z axis. Unfortunately, I couldn't determine from the code how it arrives at this number. My Z max is 310 and Z min is -5. Could you test whether the adjustment of the Z height works for you? P.S.: I'm using Klipper as the configuration.

**To Reproduce**
Steps to reproduce the behavior:
Start AUTO_SPEED Skript

**Printer (please complete the following information):**
 - Printer: Voron 2.4
 - Kinematics CoreXY
 - [ ] klippy.log attached
[image]

Anonoei, 2024-03-07:
@Fragmon this is default behavior currently. It goes to the center of your Z-axis for the tests (-5 + 310 = 305/2 = 152.5)
Margin only changes X/Y. I can add back in using a different Z height, where it will default to the center. Previously it used a Z of 10 and I didn't like it being an arbitrary value - so I figured the center would give the most general/accurate results

Anonoei, 2024-03-07:
@Fragmon This should now be fixed in [v0.3.4](https://github.com/Anonoei/klipper_auto_speed/releases/tag/v0.3.4). Feel free to re-open if you have any issues!

I'll keep your enhancement issue open until that gets implemented

closed 2024-03-07 by Anonoei