Anonoei/klipper_auto_speed #19: [ENHANCE] Im wondering if you would be able to integrate a ROBE_ACCURACY for X & Y endstops type of functionality

PhilBaz, 2024-03-06:
Please see here.
https://klipper.discourse.group/t/fr-probe-z-accuracy-or-probe-accuracy-equivalent-function-for-x-y-endstops/13986

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

XY homing accuracy can affect Z-offset accuracy with a dedicated z-homing switch that uses a pin, like with the voron 2.4. 
It would also be good for diagnosing failing X&Y switches that have a shorter lifespan in heated enclosures.

For some type of setups X & Y accuracy can be more or less important. For instance with my voron the z-endstop accuracy is a product of hitting the pin dead in the center, so is then further reliant on XY accuracy.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Equivalent of PROBE_Z_ACCURACY or PROBE_ACCURACY function for X & Y endstops.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Doesn't seem to be any, I thought maybe a macro. Someone on klipper discourse suggested I look at your work.

**Additional context**
Add any other context or screenshots about the feature request here.

Anonoei, 2024-03-07:
@PhilBaz if I understand correctly this is kind of something that already exists in AUTO_SPEED, where it measures the variance of the X/Y endstops. Is there something missing from that? 
Are you requesting for that to be exposed so you can run it independently?

PhilBaz, 2024-03-08:
@Anonoei 

Yes, exposing it to be run independently would be great. In order to quickly check the variance on the X & Y endstops.

I don't know if you are familiar with PROBE_Z_ACCURACY or PROBE_ACCURACY. But for instance I use PROBE_Z_ACCURACY with the z-endstop of my Voron 2.4, it hits the endstop 20 times and I can see the z-height at which the endstop triggers each time, then it calculates the mean, standard deviation, etc.

One nice thing about PROBE_Z_ACCURACY is that you can call it with variables like:
PROBE_Z_ACCURACY PROBE_SPEED=10 LIFT_SPEED=15 SAMPLES=20 SAMPLE_RETRACT_DIST=3 

This is very nice if you want to see how speed affects the result, or run a short test vs a long test.

I hope that makes sense :)

Anonoei, 2024-03-08:
@PhilBaz that makes sense!
I'll look how klipper implements it for the Z axis to see if there is something similar on X/Y. Since another user requested different homing speeds, I'll need to make something to control it for that anyway, which I can also add to the variance checking. This seems like it could actually be a pretty useful tool!

Should I implement it as PROBE_X_ACCURACY and PROBE_Y_ACCURACY, or something different?

PhilBaz, 2024-03-08:
@Anonoei 

Ahh, fantastic. Yes, i think it would be very useful. It's such a fundamental functionality I was kind of surprised they didn't include it in native klipper function. 

Sure, those names make perfect sense. 

Ahh, very happy you like the idea. Very excited to try it out :)

PhilBaz, 2024-06-16:
Did this already make it into Auto? sorry been a while since i pulled my head out of the ground to look around :)

legend069, 2025-02-10:
> Did this already make it into Auto? sorry been a while since i pulled my head out of the ground to look around :)

yes i added it in this pr https://github.com/Anonoei/klipper_auto_speed/pull/30
@Anonoei or anyone with repo write access update the readme.md file to state it has this new support?

PhilBaz, 2025-04-16:
@legend069 

Hello, just getting around to testing this...

I'm new to auto speed, and id just like to run X & Y end-stop accuracy test. If I install the script then I can just run the command X_ENDSTOP_ACCURACY, and it will use my default homing speeds? 

PS. thank you!

PhilBaz, 2025-04-16:
@legend069 Or do i structure it like

X_ENDSTOP_ACCURACY ACCEL=6000 VELOCITY=250 SCV=10 VALIDATE_ITERATIONS=20

6000 and 250mm are my default homing access and speed.  What is SCV= ?

legend069, 2025-04-16:
thanks for reminding me about this, i started to make a 'better' version of the xyz endstop accuracy that tested it better but forgot about it 😅 

>  and it will use my default homing speeds?

yes it will use the the values defined in the config.

> do i structure it like

``X_ENDSTOP_ACCURACY``
you can define a SAMPLES value if you'd like otherwise it would use the `default 10`
> What is SCV= ?

 SCV is Square corner velocity, it should use the default value in the config.

PhilBaz, 2025-04-17:
@legend069 

Awesome.... Thanks for your work on this! I think its a very useful tool that would help more people if they knew about it. :)

PhilBaz, 2025-04-30:
@legend069 Worked great! Thank you!   Time to change my x endstop!!!