Anonoei/klipper_auto_speed #27: [ENHANCE] Disable Motor after finishing all movements

shubhracc, 2024-08-13:
**Is your feature request related to a problem? Please describe.**
My motors are enabled after running this test and it causes them to heat up quite a bit. I have to manually disable them which is a bit of a hastle

**Describe the solution you'd like**
Enter the M18 gcode command to disable all steppers at the end of the gcode of this Macro.

**Describe alternatives you've considered**
An alternative I use is manually disabling the motors via the Mainsail UI or through KlipperScreen

**Additional context**
N/A

bistory, 2024-08-15:
Why don't you create a macro that will call M18 after klipper auto speed ?

shubhracc, 2024-08-17:
That's exactly what I did. I just thought it was inconvininent that I had to do that so I made a reccomendation so that other people who might want to use your macro have an improved user experience.

thetic, 2024-08-17:
I do not want this behavior, so if it is added please make it optional

shubhracc, 2024-08-17:
That sounds like a great idea. It could be placed in the cfg portion.