Anonoei/klipper_auto_speed #28: [BUG] When running 'AUTO_SPEED' I get an error stating axises need to be homed

courtross2010, 2024-08-24:
I installed the Klipper Auto Speed plugin on my Voron V2.4 and updated all the environments and packages.  When I run the 'AUTO_HOME' macro from the console, I get the error: 'Printer must be homed first! Found 2 homed axes.'

All of the axes have been homed.  This includes the ERCF MMU that I have on my printer.
[image]

**To Reproduce**
Steps to reproduce the behavior:
 1. Home all of the axes of the printer
 2. Open the console and type 'AUTO_HOME'
 3. Press Enter

**Printer (please complete the following information):**
 - Voron V2.4
 - Kinematics: CoreXY
 - Klipper Version: v0.12.0-286-g81de9a86
 - klippy.log attached: [klippy.log](https://github.com/user-attachments/files/16738138/klippy.log)
 - Klipper Auto Speed Version: v0.3.4-0-ga5ee2fb7

ItsAzuur, 2024-09-23:
Same issue here on my Delta. All axis are homed but the macro tells me 0 axis are homed. Other macros needing home works no issue.

theimmortal68, 2025-06-18:
I am also experiencing this issue. I have an Extended CoreXY configuration (https://github.com/SnakeOilXY/klipper-extended-corexy-kinematic) and it is telling me there are zero axis homed.

cmarchand0002, 2026-08-31:
This may be a silly question, but where do i go to disable to endstop phases?

Matze0811, 2026-09-11:
> This may be a silly question, but where do i go to disable to endstop phases?

Put in in your printer.cfg

stahlsau, 2026-10-22:
can confirm the solution of danny83 is working. Thx!