Anonoei/klipper_auto_speed #29: [BUG] Z axis might not work well with tap

gyscos, 2024-10-23:
**Describe the bug**
Running `AUTO_SPEED AXIS=Z` on my voron 2.4 with tap sensor, it appears that all "missed steps" checks reported with 0.00 or 0.03. As a result it recommended the highest possible speed, which sounds unrealistic.

**To Reproduce**
Steps to reproduce the behavior:
1. Get a voron 2.4 with tap sensor
2. Run `AUTO_SPEED AXIS=Z`
3. Observe very optimistic recommendations 

**Printer (please complete the following information):**
 - Printer: Voron 2.4
 - Kinematics Flying-gantry CoreXY
 - [ ] klippy.log attached

**Additional context**
Add any other context about the problem here.

AlatusAles, 2024-11-01:
Вверх!
Я получаю нереалистичные значения.
[image]

GavinHanson, 2024-11-18:
Can confirm, I'm also seeing this same result.