Anonoei/klipper_auto_speed #33: klipper_auto_speed uninstall script.

racoutlaw, 2024-12-04:
Installed this but does not work well on my printer.

Need an uninstall script so that i can uninstall this from klipper and not have to worry about breaking something by just deleting it from directory.

Anonoei, 2024-12-05:
@racoutlaw I should probably add that
At the time I didn't make an uninstaller because you can remove it from your printers config where you define [klipper_auto_speed]
By removing (or commenting) that line, klipper doesn't load the plugin

racoutlaw, 2024-12-06:
> @racoutlaw I should probably add that At the time I didn't make an uninstaller because you can remove it from your printers config where you define [klipper_auto_speed] By removing (or commenting) that line, klipper doesn't load the plugin

Yea other than it is giving me this error is why 
klipper
v0.12.0-316-g96cceed2
Repo has untracked source files: ['klippy/extras/auto_speed.py']