Anonoei/klipper_auto_speed #34: [BUG] Newest changes causing Internal error on command:"G28"

livinhack, 2024-12-05:
**Describe the bug**
I updated aut_speed today
After that, my printer (sensorless homing) errors out after the first axis.

**To Reproduce**
Steps to reproduce the behavior:

Latest auto_speed + sensorless homing

**Printer (please complete the following information):**
 - Printer: Voron 2.4
 - Kinematics: CoreXY

**Additional context**
Deactivating the plugin (#) solves it...

```
Internal error on command:"G28"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/v24/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/v24/klipper/klippy/webhooks.py", line 436, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/v24/klipper/klippy/gcode.py", line 230, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/v24/klipper/klippy/gcode.py", line 212, in _process_commands
    handler(gcmd)
  File "/home/v24/klipper/klippy/extras/homing_override.py", line 61, in cmd_G28
    self.template.run_gcode_from_command(context)
  File "/home/v24/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/v24/klipper/klippy/gcode.py", line 227, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/v24/klipper/klippy/gcode.py", line 212, in _process_commands
    handler(gcmd)
  File "/home/v24/klipper/klippy/extras/homing_override.py", line 23, in cmd_G28
    self.prev_G28(gcmd)
  File "/home/v24/klipper/klippy/extras/homing.py", line 281, in cmd_G28
    kin.home(homing_state)
  File "/home/v24/klipper/klippy/kinematics/corexy.py", line 64, in home
    homing_state.home_rails([rail], forcepos, homepos)
  File "/home/v24/klipper/klippy/extras/homing.py", line 225, in home_rails
    self.printer.send_event("homing:home_rails_end", self, rails)
  File "/home/v24/klipper/klippy/klippy.py", line 223, in send_event
    return [cb(*params) for cb in self.event_handlers.get(event, [])]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/v24/klipper/klippy/klippy.py", line 223, in <listcomp>
    return [cb(*params) for cb in self.event_handlers.get(event, [])]
            ^^^^^^^^^^^
  File "/home/v24/klipper/klippy/extras/autospeed/main.py", line 126, in handle_home_rails_end
    second_homing_speed = float(config["second_homing_speed"])
                                ~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'second_homing_speed'
Dumping stepper 'stepper_x' (mcu) 6 queue_step:
queue_step 0: t=6966503922 p=10069 i=62500 c=320 a=0
queue_step 1: t=6986503922 p=10389 i=62500 c=320 a=0
queue_step 2: t=7006503922 p=10709 i=62500 c=320 a=0
queue_step 3: t=7026503922 p=11029 i=62500 c=320 a=0
queue_step 4: t=7046503922 p=11349 i=62500 c=320 a=0
queue_step 5: t=7011694958 p=10764 i=0 c=0 a=0
Dumping stepper 'stepper_y' (mcu) 6 queue_step:
queue_step 0: t=6966503922 p=-10069 i=62500 c=-320 a=0
queue_step 1: t=6986503922 p=-10389 i=62500 c=-320 a=0
queue_step 2: t=7006503922 p=-10709 i=62500 c=-320 a=0
queue_step 3: t=7026503922 p=-11029 i=62500 c=-320 a=0
queue_step 4: t=7046503922 p=-11349 i=62500 c=-320 a=0
queue_step 5: t=7010965648 p=-10764 i=0 c=0 a=0
Dumping trapq 'toolhead' 1 moves:
move 0: pt=15.849567 mt=1.926760 sv=40.000000 a=0.000000 sp=(0.000000,-174.733333,0.000000) ar=(0.000000,1.000000,0.000000)
Requested toolhead position at shutdown time 17.533638: (0.0, -107.3705030666667, 0.0)
Stats 29063.4: gcodein=0  mcu: mcu_awake=0.000 mcu_task_avg=0.000001 mcu_task_stddev=0.000001 bytes_write=11717 bytes_read=15538 bytes_retransmit=9 bytes_invalid=0 send_seq=786 receive_seq=786 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=400037013 toolhead: mcu_awake=0.056 mcu_task_avg=0.000048 mcu_task_stddev=0.000119 bytes_write=15096 bytes_read=13453 bytes_retransmit=0 bytes_invalid=0 send_seq=764 receive_seq=764 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=64000733 adj=63994653 host: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_stddev=0.000003 bytes_write=952 bytes_read=5057 bytes_retransmit=0 bytes_invalid=0 send_seq=131 receive_seq=131 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000919 adj=49996527  heater_bed: target=0 temp=37.0 pwm=0.000 Raspberry_Pi: temp=48.0 mcu: temp=35.8 toolhead_mcu: temp=25.7 Chamber: temp=18.9 sysload=0.35 cputime=25.926 memavail=7687808 print_time=17.776 buffer_time=0.000 print_stall=0 extruder: target=0 temp=87.2 pwm=0.000
Stats 29064.4: gcodein=0  mcu: mcu_awake=0.000 mcu_task_avg=0.000001 mcu_task_stddev=0.000001 bytes_write=11723 bytes_read=15686 bytes_retransmit=9 bytes_invalid=0 send_seq=787 receive_seq=787 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=400035383 toolhead: mcu_awake=0.056 mcu_task_avg=0.000048 mcu_task_stddev=0.000119 bytes_write=15102 bytes_read=13570 bytes_retransmit=0 bytes_invalid=0 send_seq=765 receive_seq=765 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=64000736 adj=63996809 host: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_stddev=0.000003 bytes_write=958 bytes_read=5073 bytes_retransmit=0 bytes_invalid=0 send_seq=132 receive_seq=132 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000925 adj=49998214  heater_bed: target=0 temp=36.8 pwm=0.000 Raspberry_Pi: temp=47.4 mcu: temp=36.0 toolhead_mcu: temp=25.4 Chamber: temp=18.8 sysload=0.35 cputime=25.932 memavail=7687808 print_time=17.776 buffer_time=0.000 print_stall=0 extruder: target=0 temp=87.0 pwm=0.000
Stats 29065.4: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000004 mcu_task_stddev=0.000008 bytes_write=11729 bytes_read=15878 bytes_retransmit=9 bytes_invalid=0 send_seq=788 receive_seq=788 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=400033523 toolhead: mcu_awake=0.038 mcu_task_avg=0.000049 mcu_task_stddev=0.000120 bytes_write=15108 bytes_read=13705 bytes_retransmit=0 bytes_invalid=0 send_seq=766 receive_seq=766 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=64000778 adj=63997618 host: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_stddev=0.000003 bytes_write=964 bytes_read=5089 bytes_retransmit=0 bytes_invalid=0 send_seq=133 receive_seq=133 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000927 adj=49998772  heater_bed: target=0 temp=36.7 pwm=0.000 Raspberry_Pi: temp=48.0 mcu: temp=36.0 toolhead_mcu: temp=25.6 Chamber: temp=18.8 sysload=0.35 cputime=25.939 memavail=7687808 print_time=17.776 buffer_time=0.000 print_stall=0 extruder: target=0 temp=86.7 pwm=0.000
Stats 29066.4: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000004 mcu_task_stddev=0.000008 bytes_write=11735 bytes_read=16037 bytes_retransmit=9 bytes_invalid=0 send_seq=789 receive_seq=789 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=400035011 toolhead: mcu_awake=0.038 mcu_task_avg=0.000049 mcu_task_stddev=0.000120 bytes_write=15114 bytes_read=13808 bytes_retransmit=0 bytes_invalid=0 send_seq=767 receive_seq=767 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=64000773 adj=63998735 host: mcu_awake=0.000 mcu_task_avg=0.000001 mcu_task_stddev=0.000002 bytes_write=970 bytes_read=5116 bytes_retransmit=0 bytes_invalid=0 send_seq=134 receive_seq=134 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000929 adj=49999417  heater_bed: target=0 temp=36.5 pwm=0.000 Raspberry_Pi: temp=48.5 mcu: temp=35.8 toolhead_mcu: temp=25.6 Chamber: temp=18.7 sysload=0.35 cputime=25.945 memavail=7687808 print_time=17.776 buffer_time=0.000 print_stall=0 extruder: target=0 temp=86.4 pwm=0.000
Stats 29067.4: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000004 mcu_task_stddev=0.000008 bytes_write=11741 bytes_read=16181 bytes_retransmit=9 bytes_invalid=0 send_seq=790 receive_seq=790 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=400036214 toolhead: mcu_awake=0.038 mcu_task_avg=0.000049 mcu_task_stddev=0.000120 bytes_write=15120 bytes_read=13925 bytes_retransmit=0 bytes_invalid=0 send_seq=768 receive_seq=768 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes= [... cut ...] get=0 temp=36.0 pwm=0.000 Raspberry_Pi: temp=46.9 mcu: temp=35.9 toolhead_mcu: temp=25.7 Chamber: temp=18.8 sysload=0.32 cputime=25.969 memavail=7687808 print_time=17.776 buffer_time=0.000 print_stall=0 extruder: target=0 temp=85.6 pwm=0.000
Attempting MCU 'mcu' reset command
b'Got EOF when reading from device'
Attempting MCU 'toolhead' reset command
Attempting MCU 'host' config_reset command
b'Got EOF when reading from device'
webhooks client 140732196396432: Disconnected
Restarting printer
```

Anonoei, 2024-12-06:
@livinhack Will fix. Thank you!

Anonoei, 2024-12-06:
This should be fixed in the latest commit. Let me know if you still have issues!

closed 2024-12-06 by Anonoei

livinhack, 2024-12-06:
Works now. 👍